AfterEarthLTD/Solder-Reflow-Plate #2: [Suggestion] Change microcontroller to RP2040

VinnyCordeiro, 2022-02-27:
The RP2040 is a pretty capable, powerful, cheap, and most important of all, available microcontroller on these shortage times. I understand that the ATmega328P is easier to handle, and that this suggestion would also require a complete rewriting of the code, but not being able to build a reflow plate because of chip scarcity isn't exactly nice, and I don't like the prospect of removing the 328P from an existant Arduino, it's just postponing the problem.

Anyway, thank you for your project! Will build one as soon as possible. :)

adrianmihalko, 2022-02-27:
I would have similar suggestion: what about an ESP8266/ESP32 as controller? Many components are available for it by default (OLED screen, etc.) and it is widely available. For making it even simpler controlling the temperature could be possible in a simple web interface.

KnatteAnka, 2022-02-27:
Maybe even a Wemos formfactor so it can be a shield for either ESP8266 or ESP32 (ESP8266 fits in space for ESP32)

AfterEarthLTD, 2022-02-27:
Thanks all for your input. Hadn't much considered the RPI2040 or the ESP range for that matter. You'll be pleased to know I have already made a small redesign for the ATmega4809 that is/was available a few weeks ago. I placed a production order of 1000 from NextPCB that should be arriving in the next couple of days. If we continue to struggle with microcontroller availability I will certainly redesign for some more available chips! Thanks all

Hartland, 2022-03-03:
The Arduino IDE has the RP2040 in its board library. Very little coding change would be needed to use it.

Technerder, 2022-04-18:
+1 for the RP2040, its easy to use, cheap and seems to have an abundance of stock.

andywarburton, 2022-07-30:
+1 for RP2040. It can also run Arduino code these days so barely any rewriting will be required

LordPinhead, 2022-08-24:
I looked up what the new chip for V3 would cost and I nearly fell of my chair: 18 bucks on LCSC, 21 at Mouser and for others, I didn't even look. A RP2040 would be hard to solder, so why not just put on a Raspberry Pi Pico, I saw projects soldering the PCB directly onto their boards, no headers needed.

platima, 2023-02-05:
Why not use something that's much more open I think Pineseed or Pinenut would be the way to go. Overkill for sure, but less than $1 ea! https://pine64.com/product-category/internet-of-things/iot-wifi-ble/