AfterEarthLTD/Solder-Reflow-Plate #3: [Suggestion] will it be able to have an external controller version?

zero-custom, 2022-02-27:
As the video concerned, the lifetime of the reflow plate is a problem.
Thanks to the popularity of the 3d printer, there's lots of cheap heat plate on the market.
So will it be able to be an external controller for the heat plate?
[image]

AfterEarthLTD, 2022-02-27:
Nice idea. Unfortunately 3D printer plates aren't meant for the high temperatures needed to reflow solder paste. I have had the suggestion of making removable/replaceable plates though. When I know a little more about the longevity I may design this.

speedcrafter442, 2022-02-27:
The longevity of the PCBs will be the biggest issue, most PCBs will have a worktime of only tens of hours. Carl Bugeja has tried to do the same. 
https://www.youtube.com/watch?v=fbBXPeC-FPQ https://www.youtube.com/watch?v=ZChSbpBbrt4

AfterEarthLTD, 2022-02-28:
For sure. He was actually my original inspiration for this board. 
From my testing so far the boards show little to no damage, especially since they are software limited to 180C. I do hope to make some improvements including better temperature control for future versions.
Stay tuned as I'll write a test sketch to measure and record how many cycles it can withstand before damage occurs.

speedcrafter442, 2022-02-28:
Maybe a thermocouple taped to the bottom of the heater could help

VinnyCordeiro, 2022-03-03:
An alternative would be an external PTC heating plate made out of metal. Already seen some that can reach up to 250°C.