AfterEarthLTD/Solder-Reflow-Plate #4: Request: Add schematics

MarkusvonStaden, 2022-02-27:
First of all: Great Work! This is awesome!

I'd like to make a version with USB C. 
Could you upload the schematic in EasyEDA format? Currently the schematic is only available as PDF. (Or I am just dumb)
It would save quite some not not having to reverse draw the schematic.

AfterEarthLTD, 2022-02-27:
Have a look in /BoardVersions/70mmby50mmVer2.4/BoardFiles/EasyEDA
There should be a .json that you can import?
Let me know if it works as I've never done an import/export of EasyEDA before

Good luck with the USB C version by the way!! I had planned to do one myself in a few months. Will be interested to see how you get on

MarkusvonStaden, 2022-02-28:
I think this is only the PCB. At least I did not get any schematic out of it.

AfterEarthLTD, 2022-02-28:
Righto, sorted it.
Downloading the project only gave me a Readme.txt file annoyingly. 
However, I have exported the schematic in .json form and uploaded under /BoardVersions/70mmby50mm/BoardFiles/EasyEDA
Have a look, you should see both Schematic and PCB there now 
Ta

closed 2022-02-28 by AfterEarthLTD