AfterEarthLTD/Solder-Reflow-Plate #10: [Suggestion] Replace Barrel Jack by USB-C

ncx94, 2022-02-28:
I always try to build everything in my toolbox around the USB-C power supplies so I can have one power supply to rule them all  haha :)

If 100W is enough power for the board, I think that is a cool upgrade :)

AfterEarthLTD, 2022-03-01:
Certainly!
I had initially designed around a barrel jack for simplicity to begin with, but be sure there will be a USB C version in the not too distant future. 
I plan to work on one myself and I have heard mention of some others forking off the repo and doing the same. Long live open-source!

ncx94, 2022-03-01:
Awesome! In the mean time, I ordered everything to build 3 of then rigth now, I'll rebuild one when USB-C comes in haha.

I would suggest the STUSB4500 chip which would be perfect for that, but unfortunately in these times they are unavailable... 

Anyway in my opinion, add USB-C and a replaceable heating part, and you have an amazing product :)

MarkusvonStaden, 2022-03-03:
FYI:
I am starting right now to build a USB C Version. But it might take some time to get it finished, since it is my first real project.
It is not as easy as I initially thought, since USB C PD is designed for 3A, but according to specification it should also handle 5A.
USB C doesn't have a 12V Level, only 15V, so we get a higher current. I will have to do some testing...
I think, I might use the 15 or 20V line and a step-down circuit.

PS: Maybe we could do a discord channel or something, where we can communicate with each other?

martinlanza, 2022-03-03:
I'm not sure USB is a good option

Most notebooks these days run off 20v - banggood has adaptors for all of
them to a barrel plug - would be handy to be able to use the notebook PSU

ncx94, 2022-03-03:
Awesome @MarkusvonStaden !
I would also make the electronics and heating plate separate as well, while being at it haha!
Tell me if I can help you.

Regarding 12V, yes not all usb-c pd supplies have that option, don't think it's actually part of the standard. However that shouldn't be a problem in my opinion, we could probably make it work regardless of the power input (in a limited range, like 12-20v) by measuring the actual Vin and adapting the pwm settings around the input voltage.

Totally in for the discord channel!

@martinlanza In my opinion, USB C is way more common nowadays than barrel connectors.. However I think we could use both options and combine them on the pcb so anyone can choose the preferred one

axoulc, 2022-03-03:
Hi, I started to fork this project in order to make USB-C board with STUSB4500 and an other mcu (STM32G030F). The Schematic is already done. We can work together if you want.
Sincerely

ncx94, 2022-03-03:
> Hi, I started to fork this project in order to make USB-C board with STUSB4500 and an other mcu (STM32G030F). The Schematic is already done. We can work together if you want. Sincerely

Awesome! :)
The STUSB4500 is definitely a top notch option, but pretty hard to get these days unfortunately! Why did you change the MCU?

If we can get the board with both USB-C and Barrel Jack, external heater board, and maybe profile automatic heatup, I think it's awesome

axoulc, 2022-03-03:
> Awesome! :) The STUSB4500 is definitely a top notch option, but pretty hard to get these days unfortunately! Why did you change the MCU?
>
> If we can get the board with both USB-C and Barrel Jack, [...]

I prefer to change the mcu to have availability in the near future. This component is smaller, more powerful and cheap.
Keep the Barrel Jack will not be a problem. 
For USB PD IP2721 can also be a solution.

ncx94, 2022-03-03:
I see :)
There have been multiple discussions around the MCU change, mostly because of the Atmega328p availability, I think it would be great if everybody settled on one choice so we can all go in the same directions hahaha

Yes, IP2721 does work too :)
I'd rather use brand name parts though, had too much issues with performance and availability in the past with obscure components haha

axoulc, 2022-03-03:
My choice of MCU will not be objective I am a  STM32 fan  ;)

ncx94, 2022-03-03:
Haha don't worry we're in the same boat here haha! 
But sometimes overkill isn't the best way to kill hahah

MarkusvonStaden, 2022-03-03:
Oh, I see, we get quite a big community here :D
I used the IP2721 in my Design, since I have a bunch of them around.
I plan to finish it today, so I can order the first PCBs to test.

AfterEarthLTD, 2022-03-04:
Wow guys! 
This sounds great. I love the idea of making it USB-C powered. 
You'll certainly need a well specced USB PD capable of 60W or more at 20V for the same size board. Also, consider increasing the capacitance to even out the current draw when the MOSFET is being pulsed. 
Look forward to seeing what you come up with.

Also, I've created a Discord for SR Plates. https://discord.gg/T9dxeRmY Please feel free to join send invites, create a channel and manage your project through that if you like.

Alchme, 2022-03-14:
First comment on the project. Just wanted to add the new Discord link that never expires here: 

https://discord.gg/YzhG6FcCRA

AfterEarthLTD, 2022-03-14:
Done, well spotted, thanks

Johaan1314, 2022-05-30:
Hi, 

Added my implementation of IP2721_MAX12. I tried to use back similar components to reduce BOM list. Any comments?

[image]