AfterEarthLTD/Solder-Reflow-Plate #11: Code recurses

bhelterline, 2022-02-28:
I took a brief look at your software and it appears that your main_menu() function calls itself in several places under the proper circumstances.  I would suggest your model your code like a state machine instead.  These small MCUs do not have the resources to recurse without overrunning the stack and crashing.

AfterEarthLTD, 2022-03-01:
Good point!
This is my first large-scale program so I'm not surprised there is something wrong.
I'm familiar with state machines so I will look into converting it.
Thank you

bhelterline, 2022-03-01:
I also noticed that your chosen MOSFET is not a "logic level" version but you are driving it directly from the arduino pin.  This means the MOSFET will not be fully on with a Vgs = 5V.  Much better to choose a better MOSFET.  Search for "logic level MOSFET" and the datasheet needs to specify Rds or Ron at Vgs = 5V, not 10V.

AfterEarthLTD, 2022-03-02:
Nice! To be honest, I wasn't aware Logic level MOSFETs existed. I simply checked that the MOSFET I chose had a sufficiently small Vgs (1.2v min, 2.5v max). Though I can see now how you can give it more gate voltage to achieve a lower internal resistance. I'll certainly implement that for later versions of Ver 3.x (Ver3.0 is already on order so that makers can actually by something!)

Thank you for your expertise!!

ShawMerlin, 2022-03-08:
Here is a good video on using Logic Level Mosfets.  It's not a one stop solution for MCU's.

https://www.youtube.com/watch?v=56p3_aORiJ0