AfterEarthLTD/Solder-Reflow-Plate #9: FR4

Rudis1261, 2022-02-28:
Looking forward to making one of these myself, is it necessary to order a higher temp grade FR4? The price seems to shoot up when you do so 😱 

Would TG150-160 be ok?

[image]

BackLogers, 2022-02-28:
I have a similar question and doubts, JLCPCB only offers the TG130. What does the durability look like? Is the temperature of 170 degrees achievable without damaging the surface?

narendok, 2022-02-28:
same i doubt, In india, pcbpower offer 130, 150, >170 option , should i go 170 or 130-140 is fine ?

AfterEarthLTD, 2022-02-28:
Hi all,
All of the boards I have ordered and tested have been TG130. I wanted to make the boards as accessible and affordable as possible, hence staying away from exotic materials and aluminium. 
The boards are limited to 180 degrees and stay there for only a short time. During my testing they show no damage, especially with the black silkscreen. 
I imagine higher TG may last longer, but is not necessary. 

Hope that helps
Chris

narendok, 2022-02-28:
Nice, Thank you so much for sharing such amazing work, I'm trying one for sure for my startup. God bless you.

Rudis1261, 2022-02-28:
@AfterEarthLTD  Awesome thanks for confirming, I am happy with going the standard as well then :grimacing:

closed 2022-02-28 by Rudis1261