AfterEarthLTD/Solder-Reflow-Plate #14: Mount components on bottom

lsadehaan, 2022-03-03:
Hi, this is a very cool project! I was just wondering, would it be possible to mount the components on the bottom and have the LCD and buttons on the side below the board. That way the top would be flat and make it much easier to place the piece you are soldering on top - especially if it is a large piece.

lsadehaan, 2022-03-03:
Perhaps separating the heating board into a separate board that can be mounted with risers on top of the controller board would be even better - then one could replace the heating board if needed or even have different size heating boards.

martinlanza, 2022-03-03:
I really like that idea - and make heating boards available as spares

lsadehaan, 2022-03-03:
A small adaptation to the current design might be possible, with solder points for adding an external/mounted heating board. Then one can use the base unit alone, or expand it with an external heating board if you like.
The current design is nice for small parts and probably many people won't need anything more initially.

AfterEarthLTD, 2022-03-04:
Hi, nice ideas! 
Putting components on the bottom and having the entire top as a heater could work very well. I think I'd have to limit the max heating temp for low temp paste to make sure it doesn't melt the control solder of damage the components over time.

The detachable board is a good idea. A few people have suggested similar ideas. I'll certainly look into it as its a popular request. 

I plan to make a larger board in the not too distant future so I'll take this all into consideration.
Thank you

lsadehaan, 2022-03-04:
> having the entire top as a heater could work very well

I wasn't really thinking about increasing the size of the heater on top since it could damage components on the bottom, but if that could work it would be interesting. The reason for moving the components to the bottom is just so you have a completely flat element you can put a larger piece on top of - lets say you have a very large piece and you want to heat the middle part of it, with the current design that's not possible. For this to work the risers would need to be flush and not stick out the top.

AfterEarthLTD, 2022-03-04:
Oh I see!
I like that. 
On the version I have purchased to sell I have sorted out grub screws that sit flush with the top, therefore opening up the sides and the bottom. But you're right, if everything was on the bottom it would be easier to flow bigger boards.

lsadehaan, 2022-03-04:
Yes, I came across your design from a youtube video from Scott ( [GreatScott video](https://www.youtube.com/watch?v=QarizoUnRfk) ) where he compared your design to a really tiny commercial unit. When he was showing how he was working with a big board for each unit it was clear that the commercial unit, even though it is much smaller, is much easier to handle since you can just place the board on top. My take-away was that the DIY version would be way superior to the commercial one, if only you could do the same.