AfterEarthLTD/Solder-Reflow-Plate #20: Using a AtMega328p from another arduino device

Rudis1261, 2022-03-29:
So after some research, I found how to set the 329p fuse to use the internal clock in case you want to repurpose an Arduino with a suitable chip like a lilypad.

You will need to hook up the Arduino on the ICSP headers, and use another Arduino programmed as an Arduino as ISP to program the target 328p board. 

Using this command you can set the fuse to tell the Arduino to use the internal clock:
```shell
~/AppData/Local/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18/bin/avrdude.exe -PCOM8 -v -patmega328p -cstk500v1 -PCOM8 -b19200 -C avrdude.conf -U lfuse:w:0xF2:m
```

The command broken down:
```shell
# The path of your avrdude.exe file 
~/AppData/Local/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18/bin/avrdude.exe
```

```shell
# The COM port where the Arduino as ISP can be found
-PCOM8
```

```shell
# Verbose output
-v
```

```shell
# The target chip
-patmega328p
```

```shell
# The programmer avr should use
-cstk500v1
```

```shell
# The baud rate as set in the Arduino as ISP sketch uploaded to the programming Arduino
-b19200
```

```shell
# The avrdude.conf, this may not be in the same directory as the command.
-C avrdude.conf
```

  -U lfuse:w:0xF2:m

doskoi, 2022-03-31:
Thanks for the useful material, this doc do the same thing in another approach https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoToBreadboard/

doskoi, 2022-04-01:
Or from the comments inside the source code

To program
 *  1) Select the following settings under (Tools)
 *      Board->Minicore->Atmega328 
 *      Clock->Internal 8MHz
 *      BOD->BOD 2.7V
 *      EEPROM->EEPROM retained
 *      Compiler LTO->LTO Disabled
 *      Variant->328P / 328PA
 *      Bootloader->No bootloader
 *  2) Set programmer of choice, e.g.'Arduino as ISP (MiniCore)', 'USB ASP', etc, and set correct port.
 *  3) Burn bootloader (to set fuses correctly)
 *  4) Compile and upload

doskoi, 2022-04-01:
I was upload without set fuse nor use minicore to set internal clock but just sodering chip on this board,
Then I'm using breadboard with DIP 16M external clock attach to XTAL1 and XTAL2 (PIN7, 8), after this I can comminucate with 328p by programmer again.

simonstrachan, 2023-08-23:
I had my boards assembled with the Atmega already on them. Is there a way to do what GreatScott did in his video with a function generator? I can not seem to be able to get the bootloader to sync. Is it only possible to do it with an external clock or taking one from an existing Arduino?

doskoi, 2023-08-29:
You need to follow the setup at above to enable built-in clock.
> Clock->Internal 8MHz

For your circumstance, you need a external clock to do this setup work. after that, you won't need external clock.