AfterEarthLTD/Solder-Reflow-Plate #22: Issue with debouncing

ViggoAvatar, 2022-03-31:
There seems to be some kind of issue with the code not properly running with the current debouncing method. in order to get the heater to heat i need to hold both buttons while it's heating, letting go will instantly cancel the process and the animation will stop where it left off, been snooping around in the code for a bit but first I'm checking whether or not i made a mistake with the hardware.

toto99303, 2022-04-17:
> There seems to be some kind of issue with the code not properly running with the current debouncing method. in order to get the heater to heat i need to hold both buttons while it's heating, letting [...]

It's not only you. I have exaclty the same issue and I thought my board has problems. Then I recreated the whole thing on a breadboard using brand new Arduino Pro Mini 3.3V 8Mhz and it's behaving exactly the same. There's definetely a problem with the code.

ViggoAvatar, 2022-04-17:
> It's not only you. I have exaclty the same issue and I thought my board has problems. Then I recreated the whole thing on a breadboard using brand new Arduino Pro Mini 3.3V 8Mhz and it's behaving [...]

Good to see im not the only one with the problem, I have a quick fix, I'll make a branch for it. It still has problems properly dealing with double press.

toto99303, 2022-04-17:
Thanks! Did you also had issues changing temperature with Up/Down buttons?
I'll try your version now.

toto99303, 2022-04-17:
Works great, thank you very much @ViggoAvatar !

vonryan123, 2022-04-21:
4809 Version tested working!
https://github.com/vonryan123/Solder-Reflow-Plate

deafloo, 2022-05-03:
@vonryan123 would you mind creating a pull request for your 4809 software?

vonryan123, 2022-05-03:
> @vonryan123 would you mind creating a pull request for your 4809 software?

I don't really know much about code nor pretend to, if I understood what a pull request was or why its needed I would.

deafloo, 2022-05-03:
well done! With this pull request, the maintainer can copy your code in his empty directory and it is much easier to find for others

JoeLutz11, 2023-06-13:
Turning the switches by 180° solved the Problem for me, the little notch on the buttons has to be on the left side.