AfterEarthLTD/Solder-Reflow-Plate #24: Concerning the PCB melting issue

wswongat, 2022-04-10:
I would like to know which material type is using in this project? Does it start to melt for long run?
I did some searching on JLCPCB and it seems that the default material of 2 layers PCB is FR-4 Standard Tg 130-140.
I know the maximum temperature setting is 180°C from the source and it should use FR-4 with Tg 155 or above.

Anyway, thanks for sharing this project!

Wilson

ShawMerlin, 2022-04-10:
PCBWay will do TG150-TG160 for no extra charge but they have a higher shipping charge than JLCPCB.

If you are in the US then OSHPark has a standard of TG180 and even though they are pricier, they have amazing quality with free shipping.

https://www.reddit.com/r/PrintedCircuitBoard/comments/9bt5ed/how_come_jlcpcb_be_so_cheap_compared_to_others/

fennectech, 2022-04-21:
In a revision. The PCB could be a removable module that plugs into the controller.  Allowing replacing the heater component of the PCB without replacing the entire board and treating the heater as a wear part

EllaBarker, 2022-06-06:
Numbers?
Other projects failed because of this issue.