AfterEarthLTD/Solder-Reflow-Plate #27: Pretty hot C2

ManCloud, 2022-04-23:
When the plate heats up the 100uF capacitor C2 gets pretty hot.

NTC probe says something around 110°C... 
This is pretty near the max temperature the cap is rated for.

I guess it's caused by reactive power die to the PWM. I measured with an oscilloscope and got drops of approx 1V @ 31.5 Khz. 

Maybe this can be matched by a Coil?

vamatis, 2022-04-23:
Noticed the same here...

Robert-Such, 2022-09-28:
I am seeing temps getting closer to 130°C for C2...

Scrapz, 2023-02-18:
I just built the v2.4 board and noticed this myself. Did anyone have a fix for this?

brettski74, 2023-02-18:
There are a number of issues with this design. One is that the input capacitance is woefully low. This results in high ripple currents into and out of the input capacitor. I wouldn't recommend trying to smooth this out with an inductor. The currents are relatively large and so the inductance you'll need will also be relatively large and likely result in large voltage spikes without adequate capacitance to absorb them. What this really needs is a *lot* more input capacitance - somewhere in the realm of 20-40x more. There's not really a way to retrofit that onto this board without a redesign of the layout.

Curiously, I think a lot of us having been building the v3.0 hardware design over the last year and the most common firmwares for that didn't have the code in there to modify the timer/counter control registers on the atMEGA4809 and bump up the PWM frequency. As a result, we were using the default PWM frequency of 490Hz. At those frequencies, the power dissipation in the input capacitor is minimal and you could leave you finger on it for the entire reflow cycle. The input capacitors aren't really doing much for you at either frequency, so I'd suggest that your best option is to move to a better design. There is one by [DerSpatz](https://github.com/DerSpatz/PCB-reflow-solder-heat-plate) that's been around for a while. If you really want to stick with the v2.4 hardware, then I'd say your best workaround is to lower the PWM frequency by commenting out the TCCR2A and TCCR2B lines (148/9) in the setup() function and make sure your power supply can handle something like 15A of load current.