AfterEarthLTD/Solder-Reflow-Plate #29: Does the OLED display require pullup resistors?

vjkolaventy, 2022-04-29:
Hi

I am unable to make the OLED display work. I noticed that there are no pull up in schematic for the V3.0 board. Can you please advise if I need to add 4.7K or 10K pullup resistors to get the OLED display to work?

Regards
Vijay

deafloo, 2022-05-03:
I have a v3 board with working OLED. No pull up resistors required, just solder the 4 OLED-pins to the PCB.

vjkolaventy, 2022-05-03:
Florian
Can you please share your code for the v3.0 board? The board worked for me randomly once after trying many times but I don’t know what actually worked. I am not even able to make just the OLED to work.
I doubted the pin configuration of MegacoreX and used the blink code on the mosfet pin as suggested by MCUdude and it worked. 
If you cannot share the code, can you please share simple code to check the OLED. 
Regards
Vijay

deafloo, 2022-05-03:
Hi
I would try following steps:
- grab an arduino and check if the oled itself is working. (follow a tutorial like https://startingelectronics.org/tutorials/arduino/modules/OLED-128x32-I2C-display/)
- check your soldering. I had to touch up the mcu several times until every pin was connected.
- plug in 12V and measure the voltage at the +5V and 0V pin on the right side. It should obviously say 5V
- flash v3 code from #32. You will see the menu as soon as you power it on

I can assure you the MegacoreX pinout is correct. I know the numbers are different from the datasheet but it uses internal addresses so just dont touch it.

Hope that helps