AfterEarthLTD/Solder-Reflow-Plate #30: [Suggestion] Replacable heating surface

fennectech, 2022-04-29:
If you have the hot plate connected to the controller board with removable bolts that conduct the power.  This would permit replacing the hot plate without resoldering the whole thing making it much more reusable.  Treating the heating surface as a wear part

fennectech, 2022-04-29:
The idea is you have two pcbs.  The controller board with several of these 
[image].  You then attach the hot plate and screw it down with these lugs and these lugs carry the power and signals to and from the hot plate PCB.  This permits basicly zero soldering to the heat plate PCB.  Raising it off the surface of the table for improved thermal isolation and an easily replacable wear part that can be ordered from any PCB manufacturer without actually replacing the controller module.

fennectech, 2022-04-29:
This would make maintaining the product MUCH cheaper and easier.  Also allowing alternative heat surface designs to be used without replacing the expensive parts like the display and controller and power regulation stuff

ChuckYYC, 2022-06-19:
This would be very good improvement in the overall design.

LordPinhead, 2022-08-24:
I throw in my 2 cents: Use a silicon heating element like 3D printerbeds have and 5mm Aluminium plates, you control the heating element with a Mosfet and multiple sensors. Aluminium is a great conductor and when you use Airplane style Aluminium (AlMgSi0.5), it wont bend or warp that easy. I've bought a 235x235mm plate for 12 Euros, the heating pad was 19 Euros, the Mosfet was like 5 Euros. Combined with this project, you have a replaceable heating bed.

This PCB is great for small repairs though, like removing an HDMI port and resolder a new one, you don't need big space for it.