AfterEarthLTD/Solder-Reflow-Plate #42: No mention of compatible solder pastes in main readme

Robert-Such, 2022-09-28:
I just built one of these reflow plates and am pretty excited about it since I have not used the reflow technique before.  My understanding is that solder pastes have different reflow curves and therefore reflow ovens will generally be programmed in order to perform against that specified curve.  On the other hand it appears that this hot plate comes set with a default reflow curve in mind however I am unsure as to what solder paste this curve belongs to.

Could we add a quick mention somewhere in the readme for some pointers on solder paste recommendations?

Johaan1314, 2022-09-29:
why don't you just enter the figures yourself to suit your solder paste? 

//Heater Control Variables
  /*  Heater follows industry reflow graph. Slow build-up to 'warmUp' temp. Rapid ascent
   *  to 'maxTemp'. Then descent to room temperature. 
   */
  //byte maxTemp; //Declared in function call
  byte maxPWM = 0.70 * maxTemp; //Temperatures (in PWM / 255) influenced by paste temperature
  byte warmUpTemp = 0.75 * maxTemp;
  byte warmUpPWM = 0.72 * warmUpTemp;
  float t; //Used to store current temperature
  float v; //Used to store current voltage
  byte pwmVal = 0; //PWM Value applied to MOSFET
  unsigned long eTime = (millis() / 1000) + (8*60); //Used to store the end time of the heating process, limited to 8 mins

Robert-Such, 2022-09-29:
I really appreciate the response but I think we are talking about two different things here.  Looking back I see that I should have clarified my original question a bit better...

Essentially I am more concerned with staying within the hardware limitations of the device (i.e. thermal limits of the pcb, components etc) rather than just changing values in software.  For example the data sheet for ChipQuik SMD291AX solder paste recommends a reflow curve that peaks at 235C.  Would this solder paste be ok to use or could it possibly damage the device or cause accelerated wear & tear?

I see the design notes in this repo mention that although 150C is officially the rated range for the LMT85LP temp sensor it should still maintain reasonable accuracy up to 200C.  Also, In the handful of times I have used my device today (with default reflow curve) it appears that the capacitor C2 is reaching 130C during operation despite being designed for temperature ranges up to 105C.  In this case would a solder paste with a reflow peak of 235C be a good idea or not really?

Johaan1314, 2022-09-29:
I see.. you can try switching to a low melting point solder if you are worried, but if you are just using it once in a blue moon (like I am), I won't be too worried about the peak temp which just lasts a very short time. 

If you are using it much more often, this PCB hotplate probably won't last. I suspect the PCB will fail and delaminate first before the components get fried.. but that's just my guess..