AfterEarthLTD/Solder-Reflow-Plate #48: Burning bootloader with Arduino gives me Device signature error

wsiffer, 2022-11-30:
I ordered these boards from JLCPCB pre-soldered with the 328p. I am trying to burn the bootloader with an arduino and have all the pins wired directly from the UNO iscp pins to the reflow ISCP pins. 

Has anyone else got this issue? 
here is an image: 
[image]

JoeLutz11, 2022-12-22:
Yes Got the same issue but don't know how tho solve it

Johaan1314, 2022-12-23:
I think you need a crystal oscillator if you are burning it for the first
time. Check out the video link in the main page.

wsiffer, 2022-12-27:
I actually found a fix by making sure to wire properly. I did blow up a chip trying to give it a clock signal. I ordered 5 and successfully programmed the next 4 after I figured out the wiring

closed 2022-12-27 by wsiffer